knezicm/pds-2022 #38: Test Issue

popovict, 2022-11-02:
Upoznavanje procesa rada u GitHub okruženju. Dodavanje testnog VHDL fajla.

knezicm, 2022-11-02:
Zadatak je u review fazi ali nisam vidio da ste otvorili _pull request_. Zadatak treba da stoji u sekciji In Progress sve dok ne bude spreman za pregledanje (tj. kada otvorite _pull request_).

popovict, 2022-11-07:
Poštovanje profesore,

Naknadno sam predala testni VHDL fajl kako bih provjerila da li sam riješila problem koji sam imala pri prošloj predaji. Bili ste u pravu, pri kopiranju SSH ključa sam izostavila njegov dio te se ta greška propagirala dalje. Sada je u redu, hvala Vam na pomoći.

knezicm, 2022-11-07:
U redu, bitno je da je sada sve kako treba. Provjera nije prosla, zato sto je workflow drugaciji za zadatke koji dolaze. U svakom slucaju, naredne zadatke radite po uputstvima i sve bi trebalo da bude ok.

closed 2022-11-07 by knezicm