knezicm/pds-2022 #79: Create eight-bit multiplier

knezicm, 2022-11-21:
Realizovati 8-bitni množač neoznačenih brojeva `a` i `b`. Rezultat množenja se dobija na 16-bitnom izlazu `res`. Za realizaciju koristiti algoritam baziran na sabiranju parcijalnih proizvoda. Pri realizaciji koristiti pomjerače i sabirače. Pokušati optimizovati rješenje tako da se smanji broj sabirača potrebnih za realizaciju. Entitet kola treba da ima sljedeći izgled:

```
entity eight_bit_multiplier is
  port (
  A_i   : in  std_logic_vector(7 downto 0);
  B_i   : in  std_logic_vector(7 downto 0);
  RES_o : out std_logic_vector(15 downto 0)
);
end eight_bit_multiplier;
```
gdje su `A_i` i `B_i` ulazni 8-bitni brojevi koje množimo, a `RES_o` njihov proizvod.

Za realizovano kolo napisati odgovarajući samoprovjerljivi (*self-checking*) *testbench* neophodan za simulaciju kola. Naziv VHDL fajla za *testbench* treba da bude u formatu *nazivDizajna_tb.vhd*.

Jovann2424, 2022-11-27:
Da li moram koristiti zasebne komponente pomjeraca i sabiraca ili mogu koristiti funkcije u glavnom programu_

knezicm, 2022-11-27:
Sabirač i pomjerač se trivijalno realizuju u VHDL jeziku. Ukoliko smatrate da možete nešto postići time, možete ih enkapsulirati u funkcije. Međutim, funkcije se obično koriste u _testbench_ fajlovima, a rjeđe u dizajnu.

knezicm, 2022-12-21:
Prebacite zadatak u In Review kada zavrsite.

closed 2022-12-23 by knezicm